pal-robotics/aruco_ros #143: usage of passing by reference in case of getting the output from a function in case return

Tanishq30052002, 2025-04-08:
Multiple functions return data types of huge size; shouldn't we use referencing in these cases?

saikishor, 2025-04-09:
Hello @Tanishq30052002!

Please provide the permalink to the code, so we have a cross-reference here.

saikishor, 2025-04-09:
If you are talking about https://github.com/pal-robotics/aruco_ros/blob/humble-devel/aruco_ros/include/aruco_ros/aruco_ros_utils.hpp, then we cannot use the references at all as they are decoupled functions. Usually, these are optimized by the compiler using Return Value Optimiazation (RTO)

Tanishq30052002, 2025-04-09:
For Eg: https://github.com/pal-robotics/aruco_ros/blob/humble-devel/aruco_ros%2Fsrc%2Faruco_ros_utils.cpp#L66

Tanishq30052002, 2025-04-09:
> If you are talking about https://github.com/pal-robotics/aruco_ros/blob/humble-devel/aruco_ros/include/aruco_ros/aruco_ros_utils.hpp, then we cannot use the references at all as they are decoupled functions. Usually, these are optimized by the compiler using Return Value Optimiazation (RTO)
> 

Ok, I was not aware of this fact

saikishor, 2025-04-09:
I'll go ahead and close this issue :)

closed 2025-04-09 by saikishor