pal-robotics/aruco_ros #144: Consistent naming for private variables of an class

Tanishq30052002, 2025-04-08:
While going through the code, I felt the need to have a naming convention for the private variables, like keeping _ (underscore at last of the each name). But found a mix of both which makes a code less readable. Although it's not a functional change but having a correct naming convention can help in better understanding,

Tanishq30052002, 2025-04-08:
[image]

saikishor, 2025-04-09:
yes, you are right. Better to have underscore in the variable naming

Tanishq30052002, 2025-04-09:
Should I consider it as a new pr?

saikishor, 2025-04-09:
> Should I consider it as a new pr?

Sure, as you prefer :)

Tanishq30052002, 2025-04-09:
@saikishor I have a few suggestions with the codebase right now. 
* Will it be okay if we create separate headers and definition files?
* The main should not run from the file in which we define the class. Instead, create a separate main file that runs the program. Although it doesn't harm the codebase right now, implementing it makes the class more independent.

Tanishq30052002, 2025-04-09:
PR https://github.com/pal-robotics/aruco_ros/pull/145

PR #145 (2025-04-10): Consistent naming for private variables of an class
While going through the code, I felt the need to have a naming convention for the private variables, like keeping _ (underscore at last of the each name). But found a mix of both which makes a code less readable. Although it's not a functional change but having a correct naming convention can help in better understanding,

Fixes #144
closed 2025-04-10 by saikishor